Sage-Bionetworks/mhealthtools #25: Add additional options in feature extraction functions to switch between DSP vs model based features

th1vairam, 2018-07-19:


philerooski, 2018-09-25:
I'm putting this here both for my sake (to organize my thoughts) and to make clear how these changes will affect how users use our code.

Before changes:
`accelerometer_features` has call sign:
```
accelerometer_features <- function(sensor_data, transformation = NA, funs = NA, 
                                   model = NA, groups = c("axis", "Window"),
                                   window_length = 256, overlap = 0.5,
                                   time_range = c(1,9), frequency_range=c(1, 25))
```

Simply calling 
```
accelerometer_features(accelerometer_data)
```
will preprocess the data (detrend, bandpass filter, filter time), window the result, add columns for jerk, velocity, displacement, apply feature extraction functions `time_domain_summary`, `frequency_domain_summary`, and `frequency_domain_energy`, and return the concatenated features.

After changes:
`accelerometer_features` is replaced by what is now called `accelerometer_features_`, with call sign:
```
accelerometer_features <- function(sensor_data, transform, extract, groups,
                                    extract_on = c("acceleration", "jerk", 
                                                   "velocity", "displacement"))
```
To do an equivalent action as above using just one call to `accelerometer_features`, a user would call:
```
accelerometer_features(
  sensor_data = accelerometer_data,
  transform = purrr::partial(
    mhealthtools:::transform_accelerometer_data,
    transformation = mhealthtools:::transformation_window(
      window_length = 256, # This window_length is for our mhealthtools:::window function
      overlap = 0.5),
    window_length = 256, # This window_length is for our mhealthtools:::mutate_bandpass function
    time_range = c(1,9),
    frequency_range = c(1,25),
    sampling_rate = mhealthtools:::get_sampling_rate(accelerometer_data)),
  extract = mhealthtools:::default_kinematic_features(mhealthtools:::get_sampling_rate(accelerometer_data)),
  groups = c("axis", "Window"))
```
A more sane user would probably write something similar to the current implementation of accelerometer_features
```
my_accelerometer_features <- function(sensor_data, window_length = 256, overlap = 0.5,
                                      time_range = c(1,9), frequency_range = c(1,25)) {
  sampling_rate <- mhealthtools:::get_sampling_rate(sensor_data)
  funs <- mhealthtools:::default_kinematic_features(sampling_rate) 
  accelerometer_features(
    sensor_data = sensor_data,
    transform = purrr::partial(
    mhealthtools:::transform_accelerometer_data,
    transformation = mhealthtools:::transformation_window(
      window_length = window_length, 
      overlap = overlap),
    window_length = window_length,
    time_range = time_range,
    frequency_range = frequency_range,
    sampling_rate = sampling_rate),
  extract = funs,
  groups = c("axis", "Window"))
```
The advantages of having a user write the above function versus the current implementation (the "before changes" implementation) include:
1. The user explicitly specifies the transform and extract functions.
2. Being able to write a function like this makes the user more likely to understand what is going on underneath the hood, thus more able to debug or incorporate their own functions into the mhealthtools framework.

The disadvantages:
1. Significantly more verbose than the current implementation.
2. Understanding what is going on underneath the hood is not easy, and sets up a higher barrier of entry for users wanting to calculate features on their sensor data.

Advantage #1 is hurt by issues in other parts of our code. Suppose a user wanted to use a different filter than bandpass to clean up their sensor data. They would need to modify `preprocess_sensor_data`, which is called by `transform_kinematic_sensor_data`, which itself is called by `transform_accelerometer_data`. They would need to write a lot of redundant code just to change the preprocessing filter. Refactoring functions like `transform_kinematic_sensor_data` should mitigate this problem, but with so much going on in a single function call (`accelerometer_features`) it can be difficult to strike a balance between usability, comprehensibility, and flexibility.

philerooski, 2018-09-25:
Revisiting this comment this morning, I see I need to walk it back a bit.

1. `accelerometer_features_` is written for the input/transform/extract process, and has no "model" functionality as is.
2. We could potentially follow the example of sklearn for some arguments to `accelerometer_features`. For example, choosing the transformation function. Right now we have two options, 'window' and 'imf'. If these are _standard enough_ transformations to apply to data, we could write an `accelerometer_features` function which has a parameter `transformation` that takes a string (one of 'window', 'imf', or a vector of both to be applied sequentially), or a callable. See the `metric` argument for the sklearn NearestNeighbors classifier for a similarly treated function parameter http://scikit-learn.org/stable/modules/generated/sklearn.neighbors.NearestNeighbors.html#sklearn.neighbors.NearestNeighbors

th1vairam, 2018-10-04:
Fixed by #47

PR #34 (2018-08-31): Model based features
- accelerometer_features, gyroscope_features, and get_walk_features now include a `model` parameter that allows the user to pass their own feature extraction function, as long as it accepts the standard format of `sensor_data` as input.
- Sample model included in new file, `models.R`.

Resolves #25
PR #47 (2018-10-04): Models parameter propagated to all *_features functions
Resolves #25 

From the issue:

> Changes made in the latest commit:
> 
> 1. All `*_features` functions have a `models` parameter for passing a list of models.
> 2. `transform`-like, `extract`-like and `models` parameters are assigned default values of `NULL`.
> 3. `*_features` functions no longer return a single dataframe, since there is now the potential to return features from both standard feature extraction functions and model based functions. Standard feature extraction (`extract`) results are stored to a named list under `extracted [... cut]
closed 2018-10-04 by th1vairam