Sage-Bionetworks/mhealthtools #27: statcomp package dependency fails for R 3.5 onwards

th1vairam, 2018-07-19:
@itismeghasyam you need to find an alternative for permutation entropy stat comp package is not in CRAN anymore. Check this link https://cran.r-project.org/web/packages/statcomp/index.html

itismeghasyam, 2018-07-19:
```
permutation_entropy = function(probDist){
  ## This is the R package statcomp implementation of permutation_entropy
  ## Was re-written based on statcomp open code as statcomp is now not hosted on CRAN
  
  # Normalize so that the ratios all sum upto 1
  probDist = probDist/sum(probDist) 
  
  # Calculate the Shannon entropy of the normalized prob distribution
  H = purrr::map(probDist,
                 function(prob){
                   if(prob > 1.e-30){
                     return(-prob * log(prob))
                   }else{
                     return(0)
                   }
                 }) %>%
    unlist() %>%
    sum()
  
  # Divide the calculated shannon entropy by maximum shannon entropy to get normalized permutation entropy
  PE = H/log(length(probDist))
  
  return(PE)
}
```

^ This implementation of the permutation_entropy is exactly the someone as statcomp(by logic), only that the sapply in statcomp has been changed to purrr::map in above implementation, and in statcomp two functions permutation_entropy and its dependent function shannon_entropy are needed. Both those have been merged into one function permutation_entropy in the above implementation. Also I have verified the results from using this vs statcomp on `randu` dataset

itismeghasyam, 2018-09-13:
@philerooski Can you re-open this issue ?

th1vairam, 2018-09-14:
My specifying the remotes (like this in the DESCRIPTION file "Remotes: url::http://download.r-forge.r-project.org/src/contrib/statcomp_0.0.1.1000.tar.gz"), you shouldn't see this problem. Can one of you install the develop branch and let me know the status?

philerooski, 2018-09-21:
I just installed develop on 3.5.1 and didn't have any statcomp problems.

th1vairam, 2018-09-28:
Remotes: url::http://download.r-forge.r-project.org/src/contrib/statcomp_0.0.1.1000.tar.gz") in the DESCRIPTION solved this issue.

closed 2018-09-28 by th1vairam