Sage-Bionetworks/mhealthtools #36: Replication of DREAM mPower PDDB winning model in R

th1vairam, 2018-08-27:
Replicating the model from python theano to keras R seems to be not possible due to numerical rounding off errors. We need to discuss this in detail to figure out what will go into the final repository. 

Our options are:
- Test the final results prediction with R to see whether it is agreeable in spite of numerical differences (or)
- Re-train the model in R with the same architecture to get new weights

@philerooski Can you list the reason and a toy example you tried for @larssono to understand.

itismeghasyam, 2018-08-27:
My 2cents advice here would be to retrain the model in R with the same architecture, as I'm not so sure how sensitive (or robust) the current architecture is to numerical precision errors

th1vairam, 2018-08-28:
@itismeghasyam Assigned you the re-training in issue #37

philerooski, 2018-12-18:
This isn't feasible at the moment, and it's plausible that mhealthtools isn't the right framework for reproducing these features. mhealthtools operates on individual records whereas these features require multiple records from a single individual to reproduce. See #37 for more information.

For now, #37 subsumes this issue.

closed 2018-12-18 by philerooski