Sage-Bionetworks/mhealthtools #41: install issue

apratap, 2018-09-14:
**Describe the bug**

Failed to install from R 

**To Reproduce**
devtools::install_github("Sage-Bionetworks/mhealthtools")

**Screenshots**
```
clang -I/usr/local/Cellar/r/3.4.2/lib/R/include -DNDEBUG   -I/usr/local/opt/gettext/include -I/usr/local/opt/readline/include -I/usr/local/include   -fPIC  -g -O2  -c seewave_init.c -o seewave_init.o
clang -I/usr/local/Cellar/r/3.4.2/lib/R/include -DNDEBUG   -I/usr/local/opt/gettext/include -I/usr/local/opt/readline/include -I/usr/local/include   -fPIC  -g -O2  -c stft.c -o stft.o
stft.c:3:11: fatal error: 'sndfile.h' file not found
# include <sndfile.h> //Library for write or read sound file
          ^~~~~~~~~~~
1 error generated.
make: *** [stft.o] Error 1
ERROR: compilation failed for package ‘seewave’
* removing ‘/usr/local/lib/R/3.4/site-library/seewave’
Installation failed: Command failed (1)
'/usr/local/Cellar/r/3.4.2/lib/R/bin/R' --no-site-file --no-environ --no-save  \
  --no-restore --quiet CMD INSTALL  \
  '/private/var/folders/0k/5wwxg48s2kl93vk4nrh11pb40000gn/T/Rtmpry5W9g/devtools52f23751ed2f/Sage-Bionetworks-mhealthtools-8df1ea3'  \
  --library='/usr/local/lib/R/3.4/site-library' --install-tests

ERROR: dependency ‘seewave’ is not available for package ‘mhealthtools’
* removing ‘/usr/local/lib/R/3.4/site-library/mhealthtools’
```

**Desktop (please complete the following information):**
```
R version 3.4.2 (2017-09-28)
Platform: x86_64-apple-darwin16.7.0 (64-bit)
Running under: macOS High Sierra 10.13.5

Matrix products: default
BLAS: /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
LAPACK: /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libLAPACK.dylib

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base

loaded via a namespace (and not attached):
 [1] httr_1.3.1      compiler_3.4.2  R6_2.2.2        tools_3.4.2
 [5] withr_2.1.2     curl_3.2        memoise_1.1.0   knitr_1.20
 [9] git2r_0.21.0    digest_0.6.15   devtools_1.13.5
```

th1vairam, 2018-09-14:
This is a known issue with seewave dependency. @apratap checkout this website  http://rug.mnhn.fr/seewave/. You need the following

known issues
fftw
With Linux, you need to install the fftw3 andd fftw3-dev libraries. In the terminal do this:

sudo apt-get install libfftw3-3 libfftw3-dev 

libsndfile
With Linux, you need to install the libsndfile1 libsndfile1-dev libraries. In the terminal do this:

sudo apt-get install libsndfile1 libsndfile1-dev 

rgl
With Linux, there is sometimes a trouble installing the package rgl. In the terminal do this:

sudo apt-get install r-cran-rgl

and then try to re-install seewave from the R console.

th1vairam, 2018-09-14:
@philerooski We need a better way to package dependencies. Can you check seewave forum (https://groups.google.com/forum/#!forum/seewave.) for pre-existing solutions?

philerooski, 2018-09-20:
I tried to install on a fresh container of rocker/tidyverse and ran into the same issue as Abhi. After installing the system dependencies as in Thanneer's comment I was successfully able to install.

I also had no issue installing on macOS, though I wasn't sure how to get a "blank" default environment of macOS to test in.

I suggest adding Thaneer's instructions to the README for Linux users.

PR #45 (2018-09-24): Include seewave dependency install instructions in README
Resolves #41