Sage-Bionetworks/mhealthtools #43: recommend docs for all major functions to have self contained examples (even if with dummy data)

apratap, 2018-09-14:
Goes a long way to get more people interested and adopt

th1vairam, 2018-09-14:
@apratap Great suggestions. If you check out this markdown here https://github.com/Sage-Bionetworks/mhealthtools/blob/develop/vignettes/Introduction%20to%20mhealthtools.Rmd explains what vignettes are available or will be available in the near future. Do you suggest more than this for all the utility functions too?

apratap, 2018-09-14:
My suggestion was that help on major functions should have demo code (hopefully self-contained)

?mhealthtools::get_tapping_features -> lacks any example

th1vairam, 2018-09-14:
@itismeghasyam check @apratap suggestion. Basically, move some of the content in the vignettes to the main functions themselves using /examples tag. Check roxygen2 on how to write examples.

philerooski, 2018-12-18:
Resolved by #88

closed 2018-12-18 by philerooski