Sage-Bionetworks/mhealthtools #59: description of get_tapping_features not clear

apratap, 2018-10-26:
For a new user t,x,y will not be clear. Also, the function has taken away previous capabilities to read and shape JSON file. I understand the rationale to make it generic but since internal users are the first use cases it will be good to retain prior capabilities. Maybe allow taking JSON file besides data.frame

tap_data | A data frame with columns t, x, y, buttonid containing tapping measurements. buttonid can be from c('TappedButtonLeft','TappedButtonRight','TappedButtonNone') indicating a tap that has been classified as to the left, right or neither of those places on the screen

```
readTappingFile <- function(tappingJsonFile) {
  tryCatch({
    tapData <- jsonlite::fromJSON(tappingJsonFile)
    list(data = tapData, error = FALSE)
  }, error = function(err) {
    list(data = NA, error = TRUE)
  })
}
```

```
shapeTappingData <- function(tapData) {
    tapData <- dplyr::mutate(tapData, TapCoordinate = stringr::str_replace_all(TapCoordinate,"[{}}]", "")) %>%
      tidyr::separate(TapCoordinate,into = c("X", "Y"), sep = ",") %>%
      dplyr::mutate(X = as.numeric(X),Y = as.numeric(Y)) %>%
      dplyr::mutate(time = TapTimeStamp,buttonid = TappedButtonId) %>%
      dplyr::select(time,X, Y, buttonid)
    return(tapData)
}

```

philerooski, 2018-10-30:
@th1vairam Did you accidentally copy/paste your comments on issue #60 here? This issue is relevant to mhealthtools and should stay here. Also, please assign Megha (whom I believe was the one who included tapping functionality in this repo) or Abhi to this issue, depending on whose expertise you're looking for here.

philerooski, 2018-11-14:
I'm assigning @th1vairam since he is working on a PR to revamp some of the docstrings

philerooski, 2018-12-18:
This was resolved by #88

closed 2018-12-18 by philerooski