Sage-Bionetworks/mhealthtools #64: Fix model argument in tests

philerooski, 2018-11-07:
There are a few tests where the incorrect object type is passed to the `models` parameter. `models` should be a list of named models (like `models = list(my_custom_model = custom_model)`) rather than a single function (`models = custom_model`).

I see this error in
* `test_get_walk_features`
* `test_get_tremor_features`
* `test_get_rest_features`
* `test_get_kinetic_tremor_features`

itismeghasyam, 2018-11-07:
This is related to #65  and #66 , and would be best if changed along with them together

philerooski, 2018-11-07:
Similar in the sense they are small, easy fixes, sure. But otherwise they are completely independent issues. I can reassign myself to this issue if you like.

itismeghasyam, 2018-11-07:
I think that would be the best

PR #71 (2018-11-08): Fix errors in test model
Resolves #64
closed 2018-11-08 by philerooski