Sage-Bionetworks/mhealthtools #76: Possible reference missed ?

itismeghasyam, 2018-11-11:
```
restFeatures <- mhealthtools::get_rest_features(
+         accelerometer_data = restData$accelerometer,
+         gyroscope_data = restData$gyroscope,
+         gravity_data = restData$gravity,
+         funs = mhealthtools:::default_kinematic_features(samplingRate)
+     )
```
I got the following error when I tried to do the above for the first time on my Windows PC

```
Error in mutate_impl(.data, dots) : 
  Evaluation error: there is no package called <U+0091>fractal<U+0092>.
```

philerooski, 2018-11-12:
Can you include a reproducible example? I ran this using the sample data included with the package with a fresh install from the develop branch and had no issues.

philerooski, 2018-11-12:
Try this after reinstalling from develop:

```
restFeatures <- mhealthtools::get_rest_features(
              accelerometer_data = mhealthtools::accelerometer_data,
              gyroscope_data = mhealthtools::gyroscope_data,
              gravity_data = mhealthtools::gravity_data,
              funs = mhealthtools:::default_kinematic_features(100))
```

itismeghasyam, 2018-11-12:
Did you try it in windows?
I don't know how else to put a reproducible example out there other than paste the code that caused me the error.....

philerooski, 2018-11-12:
`restData$accelerometer` What is restData?
`funs = mhealthtools:::default_kinematic_features(samplingRate)` What is samplingRate?

It's not that important in this case because those inputs probably aren't causing the error you're seeing, but it's always good to include code that others can run (otherwise we can't help!).

Copy and paste this code block (after restarting your R session) and tell me if you still get the error:

```
devtools::install_github("Sage-Bionetworks/mhealthtools@develop")
mhealthtools::get_rest_features(
              accelerometer_data = mhealthtools::accelerometer_data,
              gyroscope_data = mhealthtools::gyroscope_data,
              gravity_data = mhealthtools::gravity_data,
              funs = mhealthtools:::default_kinematic_features(100))
```

itismeghasyam, 2018-11-12:
Gotcha! Will do again tonight 👍

philerooski, 2018-11-16:
Turns out we are calling a function from the fractal package, but it isn't explicitly referenced (like `fractal::func()`), so it never got automatically added to `DESCRIPTION`. The function is somewhere in the included feature extraction functions in `utils.R`.

th1vairam, 2018-11-19:
@itismeghasyam can you find the fractal function and explicitly reference it? open a new pull based on that.

itismeghasyam, 2018-11-19:
I found the function, we are using fractal for detrented fluctuation analysis (dfa).

We are referencing the function using fractal::dfa. We just missed specifying fractal as a required package.

https://github.com/Sage-Bionetworks/mhealthtools/blob/master/R/utils.R#660

philerooski, 2018-11-19:
Huh. I thought devtools would be smart enough to add the package to Imports if we reference it. Guess not.

closed 2018-11-19 by philerooski