Sage-Bionetworks/mhealthtools #82: Vignettes not found

th1vairam, 2018-11-14:


itismeghasyam, 2018-11-14:
I think this is because when we do a  install_github(), the parameter build_vignettes is FALSE.

We should be doing,
```
devtools::install_github("Sage-Bionetworks/mhealthtools", build_vignettes = TRUE)
```

philerooski, 2018-11-14:
@itismeghasyam Can you double check those commands are working for you? They aren't working for me. `vignette('extending')` actually gives me a `tibble` vignette.

```
> devtools::install_github("Sage-Bionetworks/mhealthtools", build_vignettes = TRUE)
Downloading GitHub repo Sage-Bionetworks/mhealthtools@master
✔  checking for file ‘/private/var/folders/46/wrl8_mw907d4hq6kcq0p_r4m0000gp/T/Rtmpu9Kmz6/remotese792a63ecce/Sage-Bionetworks-mhealthtools-38f17fb/DESCRIPTION’ ...
─  preparing ‘mhealthtools’:
✔  checking DESCRIPTION meta-information ...
─  checking for LF line-endings in source and make files and shell scripts
─  checking for empty or unneeded directories
─  looking to see if a ‘data/datalist’ file should be added
─  building ‘mhealthtools_0.1.1.tar.gz’
   
* installing *source* package ‘mhealthtools’ ...
** R
** data
*** moving datasets to lazyload DB
** preparing package for lazy loading
** help
*** installing help indices
** building package indices
** installing vignettes
** testing if installed package can be loaded
* DONE (mhealthtools)
> vignette("extending")
> vignette("modules")
Warning message:
vignette ‘modules’ not found 
> browseVignettes(package = "mhealthtools")
No vignettes found by browseVignettes(package = "mhealthtools")
```

itismeghasyam, 2018-11-14:
Something is off in the new Sage-Bionetworks@develop

Because the following works 
```
devtools::install_github("itismeghasyam/mhealthtools@develop", build_vignettes = TRUE)
vignette('extending', package = 'mhealthtools')
vignette('modules', package = 'mhealthtools')
browseVignettes("mhealthtools")
````

But when I try to build it from the current develop branch of Sage-Bionetworks, I get an error

```
devtools::install_github("Sage-Bionetworks/mhealthtools@develop", build_vignettes = TRUE)
Downloading GitHub repo Sage-Bionetworks/mhealthtools@develop
from URL https://api.github.com/repos/Sage-Bionetworks/mhealthtools/zipball/develop
Installing mhealthtools
'/Library/Frameworks/R.framework/Resources/bin/R' --no-site-file --no-environ  \
  --no-save --no-restore --quiet CMD build  \
  '/private/var/folders/ny/cx408pwj16j03x0gglc_xjfr0000gq/T/RtmpCM19Hp/devtools1442530bf326/Sage-Bionetworks-mhealthtools-cb361e1'  \
  --no-resave-data --no-manual 

* checking for file ‘/private/var/folders/ny/cx408pwj16j03x0gglc_xjfr0000gq/T/RtmpCM19Hp/devtools1442530bf326/Sage-Bionetworks-mhealthtools-cb361e1/DESCRIPTION’ ... OK
* preparing ‘mhealthtools’:
* checking DESCRIPTION meta-information ... OK
* installing the package to build vignettes
* creating vignettes ... ERROR

Attaching package: 'dplyr'

The following objects are masked from 'package:stats':

    filter, lag

The following objects are masked from 'package:base':

    intersect, setdiff, setequal, union

Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Warning in sensor_features(sensor_data = transformed_sensor_data, transform = NULL,  :
  No feature extraction function list or model list passed. Returning an empty feature list.
Quitting from lines 263-265 (modules.Rmd) 
Error: processing vignette 'modules.Rmd' failed with diagnostics:
unused argument (depressThr = 10)
Execution halted
Installation failed: Command failed (1)
```

itismeghasyam, 2018-11-14:
Also @philerooski , multiple packages might have vignettes with the same name. 
So the best would be to try this
```
vignette('extending', package = 'mhealthtools')
```

philerooski, 2018-11-15:
> Something is off in the new Sage-Bionetworks@develop

That's #89

philerooski, 2018-11-16:
The issue isn't that you don't explicitly reference the libraries you use in the vignettes when calling functions. Loading them using `library()` is fine. You just need to add the packages used to the Suggests section of DESCRIPTION, as mentioned [here](https://stackoverflow.com/questions/9899943/is-it-possible-to-use-non-imported-packages-in-a-package-vignette) and [here](http://r-pkgs.had.co.nz/description.html).

PR #101 (2018-11-16): add ggplot2 to suggests
Resolves #82
closed 2018-11-20 by philerooski