Sage-Bionetworks/mhealthtools #85: test_that fails with the following 4 issues

th1vairam, 2018-11-14:
1. Error: Generate GuanLab architechture (@test_models.R#54) 
2. Error: Load weights into GuanLab architechture (@test_models.R#63) 
3. Error: (unknown) (@test_models.R#68) 
4. Failure: Identify min, max gravity values for each window of tidy sensor data (@test_utils.R#460)

philerooski, 2018-11-16:
> Failure: Identify min, max gravity values for each window of tidy sensor data (@test_utils.R#460)

Seems to be an issue with `window_signal` rather than `tag_outlier_windows_`. @itismeghasyam Can you write tests to reproduce this case for `window_signal` and have it fail there? It's not clear to me what the expected output should be from `window_signal` given the input you create at `@test_utils.R#460`

philerooski, 2018-11-19:
Okay. Just to verify, this output from `window_signal` is correct?

```
> gravityVec <- c(rep(NA,255), 1)
> mhealthtools:::window_signal(gravityVec) %>% tail()
          1
[251,]   NA
[252,]   NA
[253,]   NA
[254,]   NA
[255,]   NA
[256,] 0.08
```

itismeghasyam, 2018-11-19:
Yes.
It is the output of a [discrete convolution](https://en.wikipedia.org/wiki/Convolution), and it is behaving as expected.

philerooski, 2018-11-19:
Referencing your test, you expect this output when passing `gravityVec` to `tag_outlier_windows_`:
```
> data.frame(window = as.character(1), max = NA, min = NA, stringsAsFactors=FALSE)
  window max min
1      1  NA  NA
```
which of your proposed solutions will give us this output? Or do you need to change the expected output from the test depending on the fix applied to `tag_outlier_windows_`?

itismeghasyam, 2018-11-19:
The `gravityVec` should be checked for `NAs` in `tag_outlier_windows_` and if found any should just output 

`> data.frame(window = as.character(1), max = NA, min = NA, stringsAsFactors=FALSE)
  window max min
1      1  NA  NA
`

itismeghasyam, 2018-11-19:
Or the best would be get the `na.rm` behavior from the user itself, expose it outside and maybe set the default to FALSE

philerooski, 2018-11-19:
> Or the best would be get the na.rm behavior from the user itself, expose it outside and maybe set the default to FALSE

We could expose it in `tag_outlier_windows`. Exposing it at the next level up (assay-level, since `tag_outlier_windows` is normally only called when passing gravity data to an assay level function) wouldn't make any sense (I can expand on this if you're interested in adding the parameter to the assay level).

> The gravityVec should be checked for NAs in tag_outlier_windows_ and if found any should just output

I'll leave it up to you. It seems to me that just because there's one timepoint with missing data, and _probably_ nothing funny -- like a rotated phone -- happened in that timepoint, we shouldn't mark the entire window max/min as NA.

itismeghasyam, 2018-11-19:
I just changed the behavior of `na.rm` in `tag_outlier_windows_` to `na.rm = F`, it now works as expected. I changed the tests also to reflect this. 

I'd leave it to you to implement exposing the parameter to the user at `tag_outlier_windows`, also even if there is one missing data point we SHOULD tag max/min for that window as NA. It's fine, people will do their own NA handling or input na.rm to our functions.

PR #100 (2018-11-16): fix dataset reference and model tests
Partially resolves #85
closed 2018-12-07 by th1vairam