Sage-Bionetworks/mhealthtools #94: Change the walking feature extraction to previous mPowertools pipeline

apratap, 2018-11-15:
As per discussion with Larsson and Megha we need to revert the walk task features to be extracted as per previous mpowertools pipeline.

-A

th1vairam, 2018-11-15:
I wonder what features in the past tool that is not present in the current version? @itismeghasyam can you do a diff of features and let us know whether this is worth doing this now?

th1vairam, 2018-11-15:
Also, we can make this as a secondary feature extraction function in addition to the pre-existing ones like time_domain_summary and frequency_domain_summary.

th1vairam, 2018-11-15:
@larssono can you comment on why we need this?

itismeghasyam, 2018-12-24:
What should we do with this? Should we revert back or not?

philerooski, 2019-01-02:
@itismeghasyam Let's discuss in person.