Sage-Bionetworks/mhealthtools #107: time_domain_summary breaks with short inputs

philerooski, 2018-11-21:
`time_domain_summary` throws 
```
<simpleError in fractal::DFA(values, sum.order = 1): Scale minimum cannot exceed scale maximum>
```
for inputs with length less than 8, and
```
<simpleError in scale[i] <- scale[i - 1] * scale.ratio: replacement has length zero>
```
for inputs less than length 18.

This situation can arise when we are trying to extract ACF features from sensor measurements less than a second in length.

To reproduce:

```
test_vectors <- list()
for (i in 1:20) {
  random_vector <- rnorm(i)
  test_vectors[[i]] <- tryCatch({
    mhealthtools:::time_domain_summary(random_vector, sampling_rate = 100)
  }, error = function(e) {
    return(e)
  })
}
```

th1vairam, 2018-12-01:
@itismeghasyam Explain here

itismeghasyam, 2018-12-24:
The error is a result of the `fractal::dfa()` function used in our calculation of the alpha(scaling exponent) of detrended fluctuation analysis. The writeup for the function describes it as `DFA is useful for characterizing long-memory correlations in stochastic fractal time series`, and I think a signal of 17 samples does not qualify as a good candidate for measuring long-memory correlations. It is inherent to that fractal package, nothing we can do about it. I suggest we wrap that dfa in a tryCatch statement (ideally we should be doing this for all the individual features, i.e mean, median, kurtosis etc.,)

Right now we have 
```
time_domain_summary <- function(values, sampling_rate=NA) {
.
.
  features <- dplyr::tibble(
.
.
.
    dfa = fractal::DFA(values, sum.order = 1)[[1]],
    rmsmag = sqrt(sum(values^2) / length(values))) %>% # Root Mean Square magnitude
    dplyr::mutate(IQR = Q25 - Q75,
                  complexity = sqrt(
                    var(diff(diff(values) * sampling_rate) * sampling_rate) /
                      var(diff(values) * sampling_rate)))
  names(features) <- paste0(names(features), ".tm")
  return(features)
}
```

Change this to 
```
time_domain_summary <- function(values, sampling_rate=NA) {
.
.
  features <- dplyr::tibble(
.
.
.
     dfa = tryCatch({fractal::DFA(values, sum.order = 1)[[1]]},
                   error =  function(e){NA}),
    rmsmag = sqrt(sum(values^2) / length(values))) %>% # Root Mean Square magnitude
    dplyr::mutate(IQR = Q25 - Q75,
                  complexity = sqrt(
                    var(diff(diff(values) * sampling_rate) * sampling_rate) /
                      var(diff(values) * sampling_rate)))
  names(features) <- paste0(names(features), ".tm")
  return(features)
}
```

closed 2019-01-30 by philerooski