Sage-Bionetworks/mhealthtools #113: <utils> tag_outlier_windows does not perform as expected [Example included]

itismeghasyam, 2018-12-04:
Right now `tag_outlier_windows` tags any window if any one of it's gravity components(x/y/z) changes sign. This is evident in the block of code below -->

```
gr_error <- gravity %>%
      purrr::map(tag_outlier_windows_, window_length, window_overlap) %>%
      dplyr::bind_rows(.id = "axis") %>%
      dplyr::mutate(error = sign(max) != sign(min)) %>%
      dplyr::group_by(window) %>%
      dplyr::summarise(error = any(error, na.rm = T)) %>%
      dplyr::mutate(window = as.integer(window))
    gr_error$error[gr_error$error == TRUE] <- "Phone rotated within window"
    gr_error$error[gr_error$error == FALSE] <- "None"
 ```

But this might be misleading because even if we have minor variations in the gravity axis (for x or y), we will still tag it as rotated within the window, an example of this is presented below
```
t <- seq(0,1,0.01)
x <- sin(20*t)/100
y <- cos(20*t)/100
z <- 9 + cos(10*t)/100

testDat <- data.frame(t = t, x = x, y = y, z=z)
tag_outlier_windows(testDat, window_length = 20, window_overlap = 0.5)
```

You'll see that the output says phone rotated within window for all, but this is not possible, since we are simulating for all practical purposes noise (the max of x, y is 0.01, min is -0.01). So I propose that instead of checking `error = sign(max) != sign(min)`, we check for something like `error = (sign(max) - sign(min)) > rotation_threshold`, where we will define `rotation_threshold` as a parameter in m/s^2. Something like a 0.5 or 1 maybe?. Maybe we can expose the `rotation_threshold` parameter to the users if we want to

#### Feature request
Change the function to 
```
gr_error <- gravity %>%
      purrr::map(tag_outlier_windows_, window_length, window_overlap) %>%
      dplyr::bind_rows(.id = "axis") %>%
      dplyr::mutate(error = (sign(max) - sign(min))> rotation_threshold ) %>%
      dplyr::group_by(window) %>%
      dplyr::summarise(error = any(error, na.rm = T)) %>%
      dplyr::mutate(window = as.integer(window))
    gr_error$error[gr_error$error == TRUE] <- "Phone rotated within window"
    gr_error$error[gr_error$error == FALSE] <- "None"
 ```

philerooski, 2018-12-18:
This function is all kinds of screwed up as it's currently implemented. From what I understand, this was included in order to help us identify when the user might be performing the assay incorrectly. But -- as you've pointed out -- the false positive rate is through the roof.

A potential problem I see with `error = ((max) - (min)) > rotation_threshold` is that measurements with extreme tremor could be marked as a phone rotation. Another false positive, though I'm unsure if the data supports this hypothesis.

@itismeghasyam Can you think of another implementation that might be able to salvage the original intent for this function? Looking at the sample data included in the package, it seems like x and y axes are just noise and the z axis, had the phone been flipped, would have shifted from -1 to +1. Maybe we should only be looking at the z axis? But wouldn't the axes you check for outlier windows be assay specific as well?

itismeghasyam, 2018-12-24:
I understand that there might be an issue with the tremor being tagged as rotated, but it is unlikely. You have to remember that we are looking at the gravity component to see the tagging, not the accelerometer nor the gyroscope readings. And for the gravity component to give a false reading of a rotated phone when there is high acceleration, the tracking of gravity should be way off. This can happen only if we have very sharp jerk like readings (something akin to a staccato in a music score spectrogram), which is unlikely of a tremor (a phone drop on to the floor would simulate this sharp jerk like reading). 

We should look into this to figure out the threshold ranges, another way might be to estimate the rotation using the gyroscope reading(integrating the gyroscope angular velocity to get angular displacement), and to see the max deviation from the initial value (is it pi radians, or 0.5 radians) etc., and then decide on a threshold.

philerooski, 2019-01-03:
At the very least, `error = ((max) - (min)) > rotation_threshold` is more robust than the implementation we have now. I'll include the parameter in the assay level modules and pass it to `tag_outlier_windows`.