Sage-Bionetworks/mhealthtools #115: get_ewt_spectrum: The first element of the first column is always NaN

itismeghasyam, 2018-12-05:
The first element of the first column (i.e first element of the first empirical wavelet) in the output always seems to be `NaN`, here is a toy example to reproduce it

```
t <- seq(0,1,0.001)
x <- sin(20*t) + sin(4*t) + cos(12*T)
xSpec <- mhealthtools:::get_spectrum(values = x, sampling_rate = 1000, nfreq = 1024)
xEwt <- mhealthtools:::get_ewt_spectrum(spectrum = xSpec,
                                        npeaks = 3,
                                        fraction_min_peak_height = 0.1,
                                        min_peak_distance = 1,
                                        sampling_rate = 1000)
View(xEwt)
```

It should not be `NaN`, it should correspond to the zero'th frequency

philerooski, 2018-12-05:
Looking at the git logs for mpowertools, it looks like @th1vairam originally wrote that function. (It used to be called getEWTSpectrum). I don't know enough about how that function works to make the changes myself.

Do you want me to assign you or him?

itismeghasyam, 2018-12-06:
I'll take a look at it Phil

itismeghasyam, 2018-12-07:
This was solved in PR #102

closed 2018-12-07 by philerooski