Sage-Bionetworks/mhealthtools #117: window_overlap parameter is behaving strange

th1vairam, 2018-12-07:
@philerooski Looks like you have manually modified the logic somewhere in the code to reverse the actual window_signal function interpretation that is causing following issues.

The current interpretation of window_overlap is 0.01 will give you 94 windows with 10 sec signal of window_length 512 representing 99% overlap between frames and 0.99 will give you 1 window with 1% of overlap between frames. 

Assuming this logic then the window_length should be a parameter between (0,1]. However the following throws an error.

\code{
walk_ftrs <- get_walk_features(accelerometer_data, gyroscope_data, window_length = 256, window_overlap = 1)

Error in kinematic_sensor_argument_validator(sensor_data = sensor_data,  : 
  If window_overlap is set to a non-NULL value, it must be a positive real number in the interval [0, 1).
 }

Which shouldn't be.

philerooski, 2018-12-07:
Good catch. Should we flip the window_overlap so that `window_overlap` = 0.1 means that each window overlaps by 0.1, or 10% ? The current implementation is not intuitive to me at all. Why not call it `window_separation` or something similar? Right now our window "overlap" is actually a measure of how disjoint each window is.

th1vairam, 2018-12-07:
Agreed. Let's just compute the 1-window_overlap inside the window_signal function that will solve the above issue. However why it is throwing an error at 1?

philerooski, 2018-12-07:
Because I originally implemented `kinematic_sensor_argument_validator` thinking `window_overlap = 0.1` meant the windows overlap by 0.1. I'll change how the windowing function works and keep the error checking the same.

PR #137 (2019-01-03): Change behavior of window_overlap parameter
Resolves #117
closed 2019-01-03 by philerooski