Sage-Bionetworks/mhealthtools #118: Move sysdata.rda

philerooski, 2018-12-07:
This is not an R file. It is a binary file. It has nothing to do with the proper functioning of our package and everything to do with tests. Move it to the test folder.

Please see the first sentence here: http://r-pkgs.had.co.nz/r.html

itismeghasyam, 2018-12-07:
In using `devtools::use_data`, the two most important ones are making the data external to the user in `data/` folder or to make the data internal for use in the package.

The way to make data internal with the package is to store it as `sysdata.rda` in the `R/` folder itself. 

Look at the Data chapter of the same book here: http://r-pkgs.had.co.nz/data.html 

I'll take suggestions on how to tackle this, if you still are not convinced by `sysdata.rda` 😄

philerooski, 2018-12-07:
> The way to make data internal with the package is to store it as sysdata.rda in the R/ folder itself.

You can also make data internal within the package by putting it in tests.

> Look at the Data chapter of the same book here: http://r-pkgs.had.co.nz/data.html

This recommends using an `R/sysdata.R` for functions that need precomputed data tables. None of our functions use precomputed data tables -- only our tests do. Put it in tests.

From the article:
> Again, to make this data reproducible it’s a good idea to include the code used to generate it. Put it in data-raw/.

If any of those underlying functions change, or even more likely, our input data changes, all those tests that use the precomputed dataframes will break. Better to be reproducible than to save a few seconds running tests each time.

itismeghasyam, 2018-12-07:
I will remove these two tests, as discussed with Phil in the morning, and we decided to remove away the hard coded examples for now, as it forms a chicken egg problem, how do we generate the test_*file used to crosscheck against the function output? Is by using a working version of that function. So it is better that we check for the flow of the code.

So with that I'll remove `sysdata.rda` since it only contains the test_*files

PR #132 (2018-12-20): remove sysdata.rda and dependent tests
Resolves #118
closed 2018-12-20 by philerooski