Sage-Bionetworks/mhealthtools #121: get_heartrate throws lots of warnings

th1vairam, 2018-12-07:
We need to avoid the following warnings, at least for simple use cases like the following

```
heartrate_ftrs = get_heartrate(heartrate_data)
Warning messages:
1: In if (n <= 0) stop("'n' must be a positive integer") :
  the condition has length > 1 and only the first element will be used
2: In 0:n : numerical expression has 2 elements: only the first used
3: In signal::fir1(bandpass_order, bandpass_edges, type = "high", window = seewave::hamming.w(bandpass_order +  :
  n must be even for highpass and bandstop filters. Incrementing.
4: In if (n <= 0) stop("'n' must be a positive integer") :
  the condition has length > 1 and only the first element will be used
5: In 0:n : numerical expression has 2 elements: only the first used
6: In signal::fir1(bandpass_order, bandpass_edges, type = "high", window = seewave::hamming.w(bandpass_order +  :
  n must be even for highpass and bandstop filters. Incrementing.
7: In if (n <= 0) stop("'n' must be a positive integer") :
  the condition has length > 1 and only the first element will be used
8: In 0:n : numerical expression has 2 elements: only the first used
9: In signal::fir1(bandpass_order, bandpass_edges, type = "high", window = seewave::hamming.w(bandpass_order +  :
  n must be even for highpass and bandstop filters. Incrementing.
```

itismeghasyam, 2018-12-07:
I can't fix `get_heartrate` until we fix #110 , or 
I can go ahead and add a separate window function just for heartrate, as I have done here  
(https://github.com/itismeghasyam/mpowertools/blob/master/R/getHrFromJson.R)
 [the mpowertools get_heartrate works fine]

philerooski, 2018-12-18:
Unblocked by #129