Sage-Bionetworks/mhealthtools #124: Travis keeps breaking with the following two tests

th1vairam, 2018-12-07:
@philerooski can you help @itismeghasyam figure out the reason for following two issues? Thanks.

Using TensorFlow backend.
── 2. Failure: Detrend the given sensor data (@test_utils.R#221)  ──────────────
`%>%`(...) not equal to mhealthtools:::test_mutate_detrend %>% dplyr::arrange(t, axis).
Rows in x but not y: 301, 300, 298, 297, 294, 293, 292, 289, 287, 303, 285[...]. Rows in y but not x: 303, 302, 298, 294, 292, 289, 285, 283, 281, 280, 279[...]. 
── 3. Failure: Bandpass the tidy sensor data (@test_utils.R#294)  ──────────────
`%>%`(...) not equal to mhealthtools:::test_mutate_bandpass %>% dplyr::arrange(t, axis).
Rows in x but not y: 303, 302, 300, 299, 293, 290, 289, 288, 287, 286, 284[...]. Rows in y but not x: 303, 298, 297, 294, 293, 291, 289, 286, 283, 282, 281[...].

itismeghasyam, 2018-12-07:
I will remove these two tests, I discussed with Phil in the morning, and we decided to remove away the hard coded examples for now, as it forms a chicken egg problem, how do we generate the test_*file used to crosscheck against the function output? Is by using a working version of that function. So it is better that we check for the flow of the code.

philerooski, 2018-12-08:
@itismeghasyam Let's keep this issue open for now and look into the failing tests on Monday (even if it is a chicken/egg problem).