Sage-Bionetworks/mhealthtools #141: Document all code objects

philerooski, 2019-01-04:
From Travis:

```
checking for missing documentation entries ... WARNING
Undocumented code objects:
  ‘accelerometer_data’ ‘balance_data’ ‘gravity_data’ ‘gyroscope_data’
  ‘heartrate_data’ ‘kinetic_tremor_data’ ‘rest_tremor_data’
  ‘sensor_data’ ‘tap_data’ ‘walk_data’
Undocumented data sets:
  ‘accelerometer_data’ ‘balance_data’ ‘gravity_data’ ‘gyroscope_data’
  ‘heartrate_data’ ‘kinetic_tremor_data’ ‘rest_tremor_data’
  ‘sensor_data’ ‘tap_data’ ‘walk_data’
All user-level objects in a package should have documentation entries.
See chapter ‘Writing R documentation files’ in the ‘Writing R
Extensions’ manual.
```

philerooski, 2019-01-04:
@itismeghasyam Please document `heartrate_data` and `tap_data` in `R/data.R` (as described in http://r-pkgs.had.co.nz/data.html#documenting-data). I will do the rest.

philerooski, 2019-01-23:
@itismeghasyam I need some help (feel free to answer in-person)
- Is resting tremor data from holding the phone in the lap or at shoulder length?
- Is kinetic tremor data from the hand to nose assay?
- Do we need the `sensor_data` dataset? Is it from a specific assay?

itismeghasyam, 2019-01-24:
- Hand in Lap test

- Yes

- I don't think that we even need that dataset, and if we are using it anywhere it must be in the vignettes which we are rewriting. And none of functions in current mhealthtools take the whole sensor data as an input (Maybe in the future if we have a model/pipeline that takes the whole sensor data, we can include it then)

PR #159 (2019-01-28): data description for walk, balance, tremor data
Partially resolves #141
closed 2019-01-30 by philerooski