Sage-Bionetworks/mhealthtools #207: Fix references inappropriately cited

Ebedthan, 2020-02-17:
Hi,

As another issue, you should fix the below inappropriately cited references:
- Citation 1: 
The mPower study,**parkinson disease** mobile data collected using **researchkit**. should be The mPower study, **Parkinson disease** mobile data collected using **ResearchKit**.
- Citation 2: 
**NPJ** Digital Medicine should be **npj** Digital medicine
- Citation 4:
**Gent, P. van, Bruin, J. de,** Kris, & Fernandes, G. should be **van Gent, P., de Bruin, J.**, Kris, & Fernandes, G.
- Citation 6 and 7:
Same as Citation 2
- Citation 8 and 9:
Please add the issue or volume number and the DOI of the paper.
- Citation 10: The title
New sensor
and wearable technologies to aid in the diagnosis and treatment monitoring of **parkinsons disease** should be New sensor and wearable technologies to aid in the diagnosis and treatment monitoring of **Parkinson's disease**.
- Citation 11:
Same as Citation 2 and add DOI.
- Citation 12:
If there is no clear information on this paper (year of publication, source, DOI, etc.), please find another one that supports your affirmation in the text.
- Citation 13:
Change parkinson to Parkinson and add DOI.
- Citation 14:
Change parkinson to Parkinson.

Thanks.

philerooski, 2020-02-20:
I've made all changes in #208 except one:

> Citation 12:
If there is no clear information on this paper (year of publication, source, DOI, etc.), please find another one that supports your affirmation in the text.

This may not be possible. The text it supports ("The heart rate features [generated by mhealthtools] have been used as part of a validation study") is extremely specific to this tool, which, being a brand new software library, has only been used in studies internal to my organization. The cited paper is currently being drafted.

If it's acceptable to keep citation 12 as is I can merge in #208 and close this issue.

Ebedthan, 2020-02-25:
@philerooski can you find a way to precise that it is a in-house data and that it will be published soon, so we can move on?

philerooski, 2020-02-26:
Hi @Ebedthan, yes -- I added a `note` tag to the citation which now makes it a valid `@unpublished` citation according to https://www.openoffice.org/bibliographic/bibtex-defs.html

Ebedthan, 2020-02-26:
Ok, no problem so we can move forward. And end the review.

PR #208 (2020-02-26): Fix references
Fixes #207
closed 2020-02-26 by Ebedthan