Sage-Bionetworks/mhealthtools #210: [Joss Review] Paper

AKuederle, 2020-02-21:
Hey everybody,

I will collect my thoughts on the paper here (I will try not to repeat points form the other reviewers)

- Paper is well written and nicely explains the structure of the model

Issues:
- [ ] You mention that various other libraries exist. I would like to see a comparison of these packages to your module to further strengthen your statement of need.

philerooski, 2020-02-27:
This is in progress but will take a bit of time.

philerooski, 2020-03-24:
Hi @AKuederle, I've added a section at the end which compares mhealthtools to similar packages in the digital health domain. If it's satisfactory to you we can close this issue.

AKuederle, 2020-03-25:
Good from my side! Thank you for adding this!

closed 2020-03-25 by philerooski