Sage-Bionetworks/mhealthtools #214: This package cannot be installed in R 4.0

JulioV, 2020-06-27:
This package depends on `fractal` which is not available anymore in CRAN. Thus, installing mhealthrools from GitHub produces this error:

`   ERROR: dependency ‘fractal’ is not available for package ‘mhealthtools’
`

philerooski, 2020-06-30:
Hi Julio! Thanks for filing this issue.

Have you tried installing the `fractal` package using `devtools::install_github` via the [CRAN GitHub mirror](https://github.com/cran/fractal)? You will need to install the `fractal` dependencies `sapa`, `wmtsa`, and `ifultools` using `devtools::install_github` as well but once you do so you can install both `fractal` and then `mhealthtools` on R 4.0.0

JulioV, 2020-06-30:
Thanks for the pointers! I haven't tried that, what I ended up doing was removing it from [DESCRIPTION](https://github.com/Sage-Bionetworks/mhealthtools/blob/master/DESCRIPTION). At the moment I'm only working with accelerometer features and it seems the only dependency is for dfa features, right? No other features will be affected?

https://github.com/Sage-Bionetworks/mhealthtools/blob/6b14637da57829e14da40502930b8c16937f1fa8/R/feature_extraction_functions.R#L41

philerooski, 2020-07-01:
Correct. That seems to be the case.

philerooski, 2020-09-16:
Neither `fractal` nor a number of its dependencies are available on CRAN, so I've decided to remove the dfa feature in order to unblock development on mhealthtools. If you need this feature, either install an [older version of mhealthtools](https://github.com/Sage-Bionetworks/mhealthtools/commit/5a2aa003aec806d51ecc14562bb30ce482cdd5ba) or manually include this feature as part of the `funs` argument to a feature extraction function (like `accelerometer_features`). For more info on including your own feature extraction functions as part of `funs` or `models`, see `vignette("extending", "mhealthtools")`

closed 2020-09-16 by philerooski